robotology/mesh-support #1808: Head Monitor Not Working

andrearosasco, 2024-05-13:
### Robot Name 🤖

ergoCub 1.1 S/N:002

### Request/Failure description

The ergocub head monitor does not show anything

### Detailed context

The ergoCub head monitor does not show anything as if it is disconnected from `ergocub-head`. 
After restarting the robot the monitor still does not work

maggia80, 2024-05-14:
@andrearosasco next time could you log the `dmsg` output?

andrearosasco, 2024-05-14:
Yes but for a while we will not be using `ergoCub 1.1 S/N:002` . I'm gonna tag @S-Dafarra since I guess AMI will be preparing the Washington demo on it

S-Dafarra, 2024-05-14:
One interesting thing that @davidelasagna was thinking about is that, since now the monitor and the head start together, maybe at the beginning the monitor is not able to draw enough power, blocking the head from detecting it.

maggia80, 2024-05-16:
@S-Dafarra yes, there could be an issue related to the power supply. I guess it could be more related to the USB HUB. We might connect the power of the display directly to the power source of the Jetson, instead of passing through the USB HUB.

AntonioConsilvio, 2024-07-02:
Hello!

I have been monitoring this issue, and in the last month it does not seem to have reoccurred.

However, I would like to ask you for feedback if you have noticed this issue in the last period.

cc @S-Dafarra @GiulioRomualdi @andrearosasco

S-Dafarra, 2024-07-02:
I haven't noticed this either!

AntonioConsilvio, 2025-03-18:
Hi! The monitor seems working properly now! Closing!

closed 2025-03-18 by AntonioConsilvio