robotology/mesh-support #1809: Right Ankle Roll goes in over temperature

carloscp3009, 2024-05-14:
### Robot Name 🤖

ergoCub 1.1 S/N:002

### Request/Failure description

We were conducting some walking on ergoCub002, and at certain point the right ankle roll went in hardware fault due to overheating.  
There is no video attached since the failure occurs when the robot was hang so we were not recording it. 

### Detailed context

[image]

At the moment of the HF the robot was hang for the approximately 5 minutes and the motors did not seem to be warm at all.  The robot interface showed a negative value for the temperature, which most likely correspond to an error with the sensor.

It was fixed just by relaunching the robot.

### How does it affect you?

It needs to be track and consider for upcoming updates and demos.
cc @S-Dafarra

S-Dafarra, 2024-05-14:
We also had a similar issue on the left ankle. Here the log

[overheating_left_leg_nonsense.txt](https://github.com/robotology/icub-tech-support/files/15311372/overheating_left_leg_nonsense.txt)

The values seems weird:
```
 1890.253647 <ERROR> from BOARD 10.0.1.9 (left_leg-eb9-j4_5) time=1877s 830m 744u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_ankle_roll (NIB=1), Raw_temperature_value=-886)
 1895.251596 <ERROR> from BOARD 10.0.1.9 (left_leg-eb9-j4_5) time=1882s 830m 747u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_ankle_roll (NIB=1), Raw_temperature_value=-378)

@valegagge @MSECode are them particular error codes?

S-Dafarra, 2024-05-14:
We did not notice that the joint was in HF and we tried to walk. It made a weird motion causing the knee to go in overcurrent.

Anyhow, we tested it afterwards and it seemed all fine.

MSECode, 2024-05-14:
Thanks for the update since more data are good for the analysis, we know that there could still be issues on the ankles but we're gonna check all asap.

valegagge, 2024-05-15:
Hi @S-Dafarra ,
the error codes you noticed are due to errors in the temperature readings.
`-378` ==> communications errors
`-886` ==> the 2foc gets the reading error for more than 10 sec

The issue is due to the electromagnetic noise.

In any case, we could improve the documentation regarding the temperature errors.

stay tuned! 

cc @MSECode

valegagge, 2025-02-12:
[This PR](https://github.com/icub-tech-iit/documentation/pull/391) contains the promised documention regarding the raw value of the temperature.

In particular, we add the [table of error values](https://icub-tech-iit.github.io/documentation/temperature_sensors/software/dataflow/#error-handling) and a [table that let the user to transform from raw value to celsius degree](https://icub-tech-iit.github.io/documentation/temperature_sensors/software/reading_data/#table-of-typical-temperature-values).

closed 2025-12-24 by pattacini