robotology/mesh-support #1811: Left hand index and middle fingers don't calibrate

steb6, 2024-05-15:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

Two fingers of the left hand don't work

### Detailed context

During calibration, it is possible to hear a motor that keeps spinning, it looks like a cable is disconnected or something similar.
In yarpmotorgui, these fingers are idle. If I do "run", it says that velocity is zero and I cannot move them

AntonioConsilvio, 2024-05-21:
Hi @steb6, the problem was due to a broken tendon in the left distal thumb. I replaced the tendon and the robot calibrates without any problems now and the finger works!

https://github.com/robotology/icub-tech-support/assets/114915464/0d918a87-f47c-4d40-87dc-221bd256deb3

Closing!

closed 2024-05-21 by AntonioConsilvio

steb6, 2024-05-21:
Thank you @AntonioConsilvio !