robotology/mesh-support #1812: Right arm calibration issues

Ryo-Naohara, 2024-05-15:
### Robot Name 🤖

iCubLisboa01 S/N:004

### Request/Failure description

Errors in icub right hand when starting yarprobotinterface, probably due to calibration errors.
Non moving right indicator finger tip and shutdown when doing yarprobotinterface start up.

- [ ] 

### Detailed context

We got the right hand mais board to work again and appear on firmware updater, when trying to start by utilizing yarprobotinterface, whit the left arm deactivated, it seems to start the initial calibration process normally but when it gets to the hand, it starts to show some errors, the first one being "hard limit reached. The joint position is outside its hardware boundaries", it will try to close the hand but the indicator finger on the hand doesn't move, after a while it will seemingly stop and deactivate, leaning back has if it is shutting down.
What could be done to try and fix this issue, is it due to encoder calibration?

### Additional context

[log_icub-head_yarprobotinterface_1680(no left arm test).txt](https://github.com/robotology/icub-tech-support/files/15322909/log_icub-head_yarprobotinterface_1680.no.left.arm.test.txt)
cc @pli

### How does it affect you?

Can't move the robot since it doesn't finish yarprobotinterface.

AntonioConsilvio, 2024-05-23:
Hi @Ryo-Naohara, This could be caused by several factors, including a possible encoders malfunction. For this reason, I would ask you to perform the following tests:

- A video of the robot start-up during the calibration phase of the `yarprobotinterface`.

- A screenshot of the `yarpmotorgui` of the joints not working properly. 
 As some of the encoders may not be working, try moving your fingers manually and see if the encoder values move accordingly or remain frozen.

- I'm not clear if the `yarprobotinterface` is self-closing.

  In this case, you won't be able to do the previous step because you won't be able to open the `yarpmotorgui`. So, you can try going with with a terminal into the robot head (ssh icub-head), go into the robots-configuration folder (`usr/local/src/robot/robotology-superbuild/src/robots-configuration/iCubLisboa01`), edit a file called `general.xml` like this

  ````
  <group name="GENERAL">
      <param name="skipCalibration">    true   </param>
      <param name="useRawEncoderData">  false  </param>
      <param name="useLimitedPWM">      false  </param>
      <param name="verbose">            false  </param>
  ````

  Start the `yarprobotinterface` within the folder itself by giving the command `yarprobotinterface` from the terminal. 
At this point, the robot should start without performing any calibration. You should then be able to open `yarpmotorgui` (where you will see everything not-configured). 
Try moving the fingers manually and see if the encoder values move accordingly or remain frozen.

  If they remain frozen, it means that the encoder is not working.

  After this test, you can reset the `general.xml` file as it was originally.

> [!Warning]
> This last step is only necessary if the `yarprobotinterface` is self-closing and therefore the previous step is not possible.

Please, send us feedback and if you have any questions feel free to ask!

Ryo-Naohara, 2024-05-24:
Hello, yarprobotinterface is self closing.
Video of trying to run yarp robot interface below

https://github.com/robotology/icub-tech-support/assets/119933065/dae1272f-ec81-4980-9bc0-5930abfede26

Since it is self closing i tried to alter the file like you've recommended
[image]

and i've tried to run yarprobotinterface in the same folder that the general.xml was found but it seemed to crash
[image]

How can i save the log for this so that i could include it?

Ryo-Naohara, 2024-05-24:
Hello, plinio taught me how to get the log.
[log-yarprobotinterface.txt](https://github.com/robotology/icub-tech-support/files/15437006/log-yarprobotinterface.txt)

AntonioConsilvio, 2024-05-27:
Hi @Ryo-Naohara @plinioMoreno since the problems described in both issues seem to be a bit more complex than expected, we were thinking of organising a video call to try to solve the problem.

We will try to solve everything at once, but other future sessions can be organised if necessary.

As a first video call we were thinking Thursday 30 May, from 9am (Portuguese time zone) until lunchtime. 

Let us know if you will be available!

Ryo-Naohara, 2024-05-27:
Hello, I forgot that there was a holiday during 30th of May. 
Could we have a zoom meeting on Friday 31th of May? (same time).

AntonioConsilvio, 2024-05-27:
Hi! Unfortunately from the 31st until the 7th I will not be available, the first useful day can be 10th of June (but I can confirm this date next week).

Ryo-Naohara, 2024-05-27:
10th of june is a bit far away so could we do the original date of may 30th thursday?

AntonioConsilvio, 2024-05-28:
Hi, if it is more convenient we can do it tomorrow, from 1 pm to 4/5 pm (Portuguese time zone).
If tomorrow is not possible, for me the 30th is fine, but I wouldn't want to do it on one of your holidays!

Ryo-Naohara, 2024-05-28:
Hello, i think it is fine to keep it as is since we've agreed to move the holiday, on the 30th would probably be a good day for 2 reasons, one being if i am here earlier we'll have more time to diagnose icub, and the other being the lab will be emptier so it'll be better for the call itself.

Ryo-Naohara, 2024-05-28:
Hello, also for preparation for the meeting would you recommend to have any specific material ready?

AntonioConsilvio, 2024-05-29:
Hi @Ryo-Naohara, I would say that apart from the classic tools (screwdrivers, tweezers, soldering iron, multimeter) there is no need for specific tools.

We'll need the robot switched on because before physically intervening, we'll do some start-up tests to understand how to intervene.

Ryo-Naohara, 2024-06-03:
Hello, after the meeting we checked the cable on the right arm motor and it seemed like it was partially broken inside, replacing part of the cable of the motor with a new cable fixed the issue, thank you very much for the troubleshooting on thursday.

closed 2024-06-03 by Ryo-Naohara