robotology/mesh-support #1813: USB port on robot Base PC faulty behaviour

elandini84, 2024-05-21:
### Robot Name 🤖

R1 S/N:001

### Request/Failure description

The USB port on R1SN001 (CHT - Erzelli) base PC connectors panel has some strange behavior. From time to time the USB WiFi dongle we attached to it disconnects from the base PC. 

### Detailed context

We have a WiFi dongle attached to the only USB port on the robot base connectors panel. We use it to connect the robot to an external WiFi network. From the first week of April we started to see this particular behavior: without any apparent cause, the USB dongle disconnects from the PC itself (we found this out by looking at the output of `journalctl`). We tried a different model of WiFi dongle but we experienced the same exact issue. Both the WiFi dongles have been tested on R1SN003 at CRIS and this behavior could not be observed.

### Additional context

The following text files contain the output of `journalctl` in two different moments:

1. When the disconnection happened [journalctl_pre_reattach_log.txt](https://github.com/robotology/icub-tech-support/files/15390448/journalctl_pre_reattach_log.txt)
2. After removing the dongle and plugging it again in [journalctl_log_after_detach_and_reattach.txt](https://github.com/robotology/icub-tech-support/files/15390457/journalctl_log_after_detach_and_reattach.txt)

### How does it affect you?

If this issue does not get solved we cannot proceed with our joint activities with IIT_PAVIS group on RAISE project, blocking both our group and theirs on this task of the project. Given that, we do not have strict deadlines on the horizon.

elandini84, 2024-05-29:
Hi @AntonioConsilvio, the second half of June is a bit too much, since the activities that require the robot involve also another group (PAVIS). Is there really no way to move this up a little bit?

AntonioConsilvio, 2024-05-29:
Hi @elandini84, We tried to schedule an intervention earlier, but unfortunately the people needed to perform this intervention are busy with high priority activities in view of important demos. We have no availability in the first half of June.

We apologise for the wait.

AntonioConsilvio, 2024-06-18:
Hi @elandini84, We planned to perform the intervention this Thursday morning!

AntonioConsilvio, 2024-06-21:
Hi @elandini84, me and @fbiggi found the USB connector on the side of the motherboard slightly detached:

[image]

This could have caused this strange behaviour. However, the cable connection from the connector on the side of the panel was probably also weak, in fact moving it broke the cables:

[image]

We have now restored everything as it should be, and tested the robot and everything is working properly.

However, before closing the issue I am waiting for feedback from you! Thank you!

elandini84, 2024-07-01:
The robot works fine, so for me the issue can be closed. Thanks a lot @AntonioConsilvio and @fbiggi

closed 2024-07-01 by AntonioConsilvio