robotology/mesh-support #1814: False "external fault button pressed" error

paliasgh, 2024-05-22:
### Robot Name 🤖

iCubWaterloo01 S/N:044

### Request/Failure description

Hello, when we run `yarprobotinterface`, face and both arms produce errors like this, while the button is not engaged:

[image]

### Detailed context

This happened shortly after we repaired the tendons of left shoulder. Due to an accident (robot hit itself) cable 7 broke and 8 snapped out. When I turned the robot on, cable 1 broke as well, as the motor started turning in the other direction. We then followed the wiring procedure to replace broken cables and put tendons back in place. After we fixed all the tendons, we could turn the robot on (run `yarprobotinterface`) and we tested the shoulder. Everything was fine. However, next time, the robot started giving false "external fault button pressed" error. We can not run `yarprobotinterface` anymore. In fact, all other parts (legs and torso) work fine, but it does not start head and both arms due to the error.

We tested our emergency stop button with a multi meter and it was fine. Also the socket looked fine. 'FirmwareUpdater' can find all the boards.

As a side question, while we fix this, is there any way to bypass the fault button error (turn robot on without the button)?

### Additional context

Here is `yarprobotinterface` log file:
[yarprunlog_22_05_2024_16_04_27.log](https://github.com/robotology/icub-tech-support/files/15408429/yarprunlog_22_05_2024_16_04_27.log)

AntonioConsilvio, 2024-05-23:
Hi @paliasgh, the problem seems to involve the whole upperbody. 
The fault cables are the white ones, and they have a common origin, i.e. the board where the fault button and the ETH cable connects (board name: `BCB` - green 🟢 in the photo below).

[image]

The fault looks like this: the connector connected to the `BCB` board (red 🔴 arrow in the picture below), from the connector two cables start, one going to the torso and lowerbody (the one that works properly) and the other going to the upperbody (which is soldered to other cables, yellow 🟡 circle)

[image]

If you look at the wiring diagram, you can see that the one that was represented by the yellow 🟡 circle in the previous photo, is a soldering point (SP_F) which carries the fault to all the boards in the upperbody.

[image]

My opinion is that the piece of cable between the soldering point and the `BCB` connector may be damaged and therefore cause the whole upperbody to malfunction.

Please, send us feedback and if you have any questions feel free to ask!

patrickshas, 2024-05-23:
Hi, @AntonioConsilvio,
Thank you for your quick reply.
This means that we need to take off the black backpack cover? Is this correct?
Is there anything we need to watch out for while doing this?
Thanks again

paliasgh, 2024-05-23:
Hello @AntonioConsilvio ,

Thanks for your prompt response. We followed the instruction [here](https://icub-tech-iit.github.io/documentation/icub_tendons/waist_legs/?h=backpack#21-chest-back-and-back-pack-covers-dismounting:~:text=B2%20%2D%20BACKPACK%20VERSION) and removed the backpack cover. The issue was obvious when we looked at `BCB` board. The connector was loose!

Before:

[image]

After:

[image]

--

Now we have another problem! When I was trying to put the cover back, I accidentally broke one of the capacitors of `BDC` board😢 (the one that is missing here):

[image]

We somehow managed to fix this without smd soldering or even removing the board. We attached two small wires to the legs of the capacitor and then attached wires to the board. The positioning of the capacitor and wires looks safe (if nothing moves). With this, we briefly turned the robot on and tested things. Everything looked alright. The board looks like this:

[image]

For a better fix, we need to order a new capacitor (since one of the legs was broken and we are not sure the wire is properly attached to the small remaining piece) and maybe more properly attach it to the board. The first question: is this the right part?

https://www.mouser.ca/ProductDetail/Nichicon/UCD1E471MNQ1GS?qs=pB3bWBO%2FVgKbyRD%252Bkq783A%3D%3D

(470 uF, 25 VDC, UCD Series)

The second question: Can we attach the new capacitor like what we did (using wires) or do we necessarily need to take the board of and attach the capacitor with SMD (which takes more time)?

Thanks again for your help!

AntonioConsilvio, 2024-05-24:
Hi @paliasgh! Glad you found the cause of the fault problem!  I'm sorry for the snag. Unfortunately the backpack cover is a bit tricky to fit, so it can happen!

The two capacitors should be the same (having the same specifications) so you can use one of them.

I recommend (if possible) mounting the SMD capacitor directly on the board, especially as leaving it with the wires could cause problems when mounting the backpack cover.

paliasgh, 2024-05-27:
Hello @AntonioConsilvio , thanks for the advice. We managed to finally attach the capacitor to the board (with a normal solder iron, but lots of work). There was almost zero space for the backpack cover to go in around that capacitor!

All our issues are fixed now. Thanks again for your support!

closed 2024-05-27 by paliasgh

AntonioConsilvio, 2024-05-28:
Good job @paliasgh!