robotology/mesh-support #1815: Left elbow doesn't calibrate

davidegorbani, 2024-05-23:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The right elbow doesn't calibrate, when we try to calibrate it with the yarpmotorgui, it trows the following error in the yarplogger:

[image]

### Detailed context

We were trying to start the robot and all the joints calibrated but the the left elbow, then we proceeded calibrating it with the yarpmotorgui but it failed.

gabrielenava, 2024-05-23:
cc @DanielePucci

gabrielenava, 2024-05-27:
while unmounting the cover with @AntonioConsilvio and @AntonioAzocar we noted that the power supply cabl was damaged:

[image]

might be related

AntonioAzocar, 2024-05-27:
> while unmounting the cover with @AntonioConsilvio and @AntonioAzocar we noted that the power supply cabl was damaged:
> 

Fixed!

[image]

Now we can mount it on the robot and test it.

cc @AntonioConsilvio

AntonioConsilvio, 2024-05-28:
Hi @davidegorbani @gabrielenava, @AntonioAzocar intervention was helpful, but unfortunately not decisive.

So, I tried replacing the 2FOC board and tested the robot, which seems to be working properly.

However, I await your feedback before closing the issue!

gabrielenava, 2024-05-28:
thank you @AntonioConsilvio! We'll test the robot in the next days to confirm if the problem is fixed!

davidegorbani, 2024-05-29:
We tested the robot yesterday and today during the demo and on both occasions, it successfully started. Thank you @AntonioAzocar @AntonioConsilvio!

AntonioConsilvio, 2024-05-29:
Thank you for the feedback! Closing!

closed 2024-05-29 by AntonioConsilvio