robotology/mesh-support #1817: The motor in right hand runs loose

marcoaccame, 2024-05-29:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

Today I run YRI on the robot and the I heard a motor of the right hand that runs loose when the finger calibration is executing. The fingers stay close and calibration does not complete.

I don't have the log. I preferred to fault the robot and stop using it.
 

### Detailed context

I started:
- on the laptop: i started `yarp server`
- on icub-head: i executed `yarprobotinterface --config ./icub_all_no_legs.xml` directly in the src folder of robots-configuration

steb6, 2024-06-04:
I update a video as reference

https://github.com/robotology/icub-tech-support/assets/24273206/f40ac0da-5e69-4e50-9009-609c0d83b061

steb6, 2024-06-10:
I wanted to add that now also the left hand has some problems, all fingers does not complete calibration

lornat75, 2024-06-13:
This issue is becoming urgent, we cannot run demos nor using the hands of the robot

AntonioConsilvio, 2024-06-17:
Hi! I replaced the broken tendon of the right index distal with a new one. But the robot had a problem not described in this issue.

@fedeceola Opened a new [issue](https://github.com/robotology/icub-tech-support/issues/1829#event-13186592498) and we will fix it tomorrow.

AntonioConsilvio, 2024-06-21:
Hi! I tested the robot. It works properly now! 🚀

Closing! ✅

closed 2024-06-21 by AntonioConsilvio