robotology/mesh-support #1819: Left hand fingers not closing completely

giotherobot, 2024-05-30:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

After a couple hours of successful manipulation teleoperation the fingers stopeed reaching complete closure.

Even after rebooting the robot the fingers were not able to reach the lower higher limit.

### Detailed context

Last plots of the position of the affected fingers:
[image]
[image]
[image]

Screenshots of the yarpmotorgui, after the robot reboot:
[image]
[image]

### Additional context

Possibly similar to https://github.com/robotology/icub-tech-support/issues/1774

### How does it affect you?

Unable to grasp objects that were previously graspable.

Nicogene, 2025-01-21:
xref:
- https://github.com/robotology/icub-tech-support/issues/1774

AntonioConsilvio, 2025-01-27:
We have implemented a mechanical upgrade that increases the range of motion on the motor side of the fingers.

Now the fingers should properly work in the full range of motion.

I am now closing all issues related to this topic, so that should there be similar problems, we know that they originated after this important upgrade.

closed 2025-01-27 by AntonioConsilvio