robotology/mesh-support #1820: Bring the iCubBarcelona01 S/N:013 back to life.

binerystorm, 2024-05-30:
### Robot Name 🤖

iCubBarcelona01 S/N:013

### Request/Failure description

Good day,
I am currently trying to get an iCubBarcelona01 S/N:013 up and running and would really appreciate pointers  as to what documentation I should be looking at.

### Detailed context

I am currently working for Radboud University. Radboud has an old iCubBarcelona01 which has, to my knowledge, not been operational for at least five years and we have lost contact with the people who previously worked on it. I have been tasked with bringing the robot back to life, however this is proving difficult as I am not sure what the relation is between the iCub version number (e.g. iCub v1.3 or iCub v2.5) and the the iCub code names (e.g. iCubBarcelona01 or iCubDarmstadt01). And thus I do not know what documentation or wiring diagrams I should be looking at, or if the documentation is even still publicly available.
Besides this we also have a iKart S/N:001 and for this I have found no information whatsoever, so I decided to ask where I should be looking as a side note here.

Any pointers will be much appreciated. As well as any arrangements that can be made to get the documentation if it is no longer publicly available.
Thanks for all your help in advance.

pattacini, 2024-06-04:
Hi @binerystorm 

`iCubBarcelona01` is an iCub v1.3.1, according to https://icub-tech-iit.github.io/documentation/icub_versions/#currently-known-icub-versions, meaning that it is a v1.3 but sports the head v2.

> And thus I do not know what documentation or wiring diagrams I should be looking at, or if the documentation is even still publicly available.

As you can imagine, yours is a very old robot, which we no longer support:
However, you can find info about wiring at:
- https://icub-tech-iit.github.io/documentation/icub_wiring/icub1_x/

It would help if you could provide the following piece of information:
- Is the iCub able to power up?
- What is the linux distro available on the laptop? 

Software-wise, we usually recommend proceeding with an update. `iCubBarcelona01` is old and as such no longer maintained. For this reason, it is now in our [`robot-icebox`](https://github.com/robotology/robots-configuration/tree/master/robots-icebox). At any rate, in the recent past, we already helped users of robots of a similar age. To this end, you may follow the instructions we provided for `iCubManchester01`:
- https://github.com/robotology/robots-configuration/issues/613#issuecomment-1959220307

>Besides this we also have a iKart S/N:001 and for this I have found no information whatsoever, so I decided to ask where I should be looking as a side note here.

This is very problematic since we've not been dealing with an iKart device for many years.
You can still find something relevant on the deprecated documentation:
- http://wiki.icub.org/wiki/IKart

---

That being said, our warm recommendation in these cases[^1] is to consider the option of delivering the robot to us for deep HW maintenance. Do you still have the robot traveling case?

Depending on the agreement and the availability of the components we may try to refurbish it or even upgrade it.

[^1]: We did it recently for `iCubLondon01`, `iCubNancy01`, for example.

binerystorm, 2024-06-14:
Thanks for the response.

The confirmation of the version number of  our iCub as well as the wiring diagrams have been very useful. 
> It would help if you could provide the following piece of information:
> 
> Is the iCub able to power up?
> What is the linux distro available on the laptop?

iCub does power up and boot. I have also managed to connect to it via ssh.
I am currently using a windows computer which is not ideal so I am planning on dual booting it with the icub server os mensioned [here](https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-from-image/) or dual booting with ubuntu and following the instructions [here](https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-from-scratch/)

I am planning on looking into the robot-icebox next week.

> This is very problematic since we've not been dealing with an iKart device for many years.
> You can still find something relevant on the deprecated documentation:
>
> http://wiki.icub.org/wiki/IKart

the link you provided here gives the error code 403 (Forbidden). This is however not high priority as I have found out that the  iKart is not strictly needed. it would be nice to get working but for now my priority is the iCub

> That being said, our warm recommendation in these cases[1](https://github.com/robotology/icub-tech-support/issues/1820#user-content-fn-1-bc93d80c755b7991e3c9d5aa0e24a295) is to consider the option [...]

This is a much appreciated offer and we do still have the traveling case, but for now I do not think it is needed (yet). But if the need arises we will certainly take you up on your offer.

For the minute I am a bit confused about the networking structure. I understand that the icub-head (10.0.0.2) communicates with the icub-server (10.0.0.1), but I have also read things about (10.0.1.104) which is the motor ip address or somthing along those lines?

I also am unsure of how the robot should be programmed once I have calibrated it. I may not have been looking in the right places but I couldn't find anything on this in the icub documentation. Should I be reading Yarp documentation to get more info on this?

Thanks for your help and have a nice weekend

pattacini, 2024-06-16:
>the link you provided here gives the error code 403 (Forbidden). This is however not high priority as I have found out that the iKart is not strictly needed

Right! I forgot that the old documentation was hosted internally via a website service that is now considered unsafe and thus is no longer accessible outside IIT.

>For the minute I am a bit confused about the networking structure. I understand that the icub-head (10.0.0.2) communicates with the icub-server (10.0.0.1), but I have also read things about (10.0.1.104) which is the motor ip address or somthing along those lines?

Yours is an old system where the internal PC mounted on the head is a PC104 board (called `pc104` with address 10.0.0.104) rather than a COM-EXP (called `icub-head` with address 10.0.0.2).

>I also am unsure of how the robot should be programmed once I have calibrated it. I may not have been looking in the right places but I couldn't find anything on this in the icub documentation. Should I be reading Yarp documentation to get more info on this?

We use the [YARP](https://yarp.it/latest/) middleware, which is a lightweight C++ library.

binerystorm, 2025-01-27:
Hi, sorry for such a late reply.
It took quite a bit of fiddling but I got the software to compile (The fiddling was due to it being old software and there have been certain changes in it's dependencies and the c standard library). I am pretty sure I have the right version of yarp and the icub-main repo, as I found some old documentation in the wayback machine: http://web.archive.org/web/20170331231018/http://wiki.icub.org/wiki/Compilation_on_the_pc104_old
which stated the versions of the software needed. However the iCubInterface still does not work. When I try to run it I get this error:
```
###################################################################################
###################################################################################
 It has been detected that your control borads are not using the same
 CAN protocol used by iCubInterface. iCubInterface cannot continue.
 Please update your system (iCubInterface and/or your control board firmware.
 For further informationn please visit: http://eris.liralab.it/wiki/Firmware
###################################################################################
###################################################################################
```
I am fairly certain I have the right firmware after I cross referenced the information I got out of the canBusLoader with this [here](https://icub-tech-iit.github.io/documentation/icub_firmware/associated-firmware/associated-firmware/#icub-v110). So after looking through the documentation I found on the way back machine I found this [here](http://web.archive.org/web/20170706093444/http://wiki.icub.org/wiki/ICubInterface_Errors) stating to unset the ICUB_CANPROTOCOL_STRICT in ccmake, however when I do this and try to configure the flag resets. No further errors are given during configuration.

I am not sure what this flag does in it's entirety, however if all it does is set the compiler flag of the same name, I tried  that, and it didn't work either.

Do you have any proposals as to how to procede or do you think its a lost cause?