robotology/mesh-support #1821: Increase the IMU frequency over 200 Hz

davidegorbani, 2024-06-04:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

We would like to increase the frequency at which the IMU streams data over 200 Hz.

### Detailed context

We were doing some tests in order to increase the frequency at which the IMU sends data over 200 Hz, which is the current frequency at which the IMU works now. Our IMU is the Xsens MTi-680G and it is connected to the head of the robot through the serial port; we tried to increase the frequency of the IMU to 400 Hz with [mtmanager](https://www.xsens.com/hubfs/Downloads/usermanual/MT_Manager_user_manual.pdf), the GUI provided by Xsens, but it was throwing an error that was suggesting to increase the baud rate of the port because some packets were lost. We tried to increase the baud rate of the serial port, but it seems like it is not possible to increase it over 115200.

CC @gabrielenava @DanielePucci

Nicogene, 2024-06-04:
In serial communication the baud rate = bit rate = the value of the clock cycle.

I don't know how many bits is a packet sent by the IMU, but by doubling the rate probably we are reaching the limit of the bandwidth of the serial we have on the COM express (we should investigate that limit cc @maggia80 @fgarini).

@davidegorbani can you provide the size of the data you are asking to the device in bits?

maggia80, 2024-06-04:
Yes, @davidegorbani you should check how many data you want to receive and add a small % for considering the overhead of the protocol (if you don't have access to it).
Check in the datasheet which is the max baudrate for the sensor and test it outside of the robot, or with the motors switched off. There could be also errors due to noise.

davidegorbani, 2024-06-04:
> is it connected to the COM express right?

Yes, correct.

We are receiving from the IMU roll, pitch, yaw and angular velocity at 200 Hz and linear acceleration and magnetometer reading at 100 Hz; all of them are vectors of three double elements so once we receive a packet with 12 doubles and once a packet with 6 doubles so on average we computed that we get 576 bits without considering the overhead of the protocol. So if we compute the bit per second we get 576*200 = 115200.
This confirms that we have to confirm the baud rate but it seems that is not allowed for the serial port of icub-head (/dev/ttyS0)

HosameldinMohamed, 2024-06-04:
> This confirms that we have to increase the baud rate but it seems that is not allowed for the serial port of icub-head (/dev/ttyS0)

Is there a limit on the baud rate of the built-in serial port on iCub-head (/dev/ttyS0) that it can't be over 115200? Because with USB (/dev/ttyUSBx) we can increase the baud rate to higher values.

davidelasagna, 2024-06-05:
If I am not mistaken, the max baud rate for RS-232 standard is 115200. The type 10 supports also RS-485 (which should supports higher baud rates) as shown in the documentation.
I think that the cable provided for the com port has only 2 wires attached to the pins for the RS-232 T/S.
There is also the jumper configuration to check.

[image]
[image]

fgarini, 2024-06-05:
Thanks @davidelasagna for the documentation reference.
Unfortunately I didn't found any specification about the driver of RS-232 interface in the `Type 10 Mini board` user manual.
However, looking in the non-Mini version, they say:
[image]
[image]
if we are lucky and they mount the same driver the speed is configurable, you can at least try with `230400 baud`.

closed 2025-12-24 by pattacini