robotology/mesh-support #1822: "MC: AEA encoder invalid data" error in shoulder joint

paliasgh, 2024-06-04:
### Robot Name 🤖

iCubWaterloo01 S/N:044

### Request/Failure description

Hello,

This morning, our robot ran into a new problem. Almost every time, after the first movement of robot's left arm, the arm goes into HF mode with the error: "MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. (Joint=l_shoulder_yaw (NIB=2), encoderPort=0)". `yarprobotinterface` continues running with other parts.

I think it is a common issue (reported many times in the past), but looking at old issues I could not find the right solution for our robot, so I am posting a new issue. It might be another cable or connector issue.

Thanks in advance for your support.

### Detailed context

Here is a `yarprobotinterface` log file:

[yarprunlog_04_06_2024_15_01_01.log](https://github.com/user-attachments/files/15569224/yarprunlog_04_06_2024_15_01_01.log)

[image]

`yarpmotorgui` screenshot after the fault:

[image]

paliasgh, 2024-06-18:
Hello, I just wanted follow up on this issue. Thank you very much!

AntonioConsilvio, 2024-06-19:
Hi @paliasgh, sorry for the wait time. Unfortunately it's been a period with several urgencies, but we will now provide you with the proper support.

`AEA encoder invalid data` means that a joint (or more) is no longer receiving correct values from the joint sensor (AEA is the name of the sensor).

The first 3 joints of the iCub shoulders are matched, which means that if one of the 3 goes HW-fault, the other two also go HW-fault.

So you have to figure out which AEA is malfunctioning.

To do this, I recommend starting the robot. When the calibration is finished and the three joints are in HW fault, start the `yarpmotorgui` and manually move the three joints (pitch, roll, yaw).
While you are manually moving the joints look at the `encoder` info from the `yarpmotorgui` and probably one of the 3 joints will always be holding the same value even though you are moving the joint.

That will be the sensor that is not working.

Once you find out which sensor is not working, I will give you more info on how to fix it!

paliasgh, 2024-06-20:
Hello @AntonioConsilvio , many thanks!

I did the test as you suggested. I tried to move all three DOFs at the same time when the joint was in HW fault and as seen below, all encoders kept updating:

https://github.com/robotology/icub-tech-support/assets/59978262/3a125266-7ad3-49bb-be8b-72e745b96ac1

Then, I restarted `yarpmotorgui` and let the code cause the HW. This time, I recorded the instance when HW happens. It looks like Joint 0 was frozen:

https://github.com/robotology/icub-tech-support/assets/59978262/db60f52f-948f-433f-9da3-a5ce6a2630e0

Next, I again restarted the robot and this time manually moved each joint. HW did not happen this time:

https://github.com/robotology/icub-tech-support/assets/59978262/ca81e2cb-9041-4d12-8e16-d779513ba42d

The funny thing is, after that robot never went into that HW fault again! I worked with the robot for the past 45 mins and everything was okay. Can the problem be a cable that was stuck in a bad position and after I moved the joints into some awkward positions the cable was released? I will keep you posted if the issue happens again, but for now, it seems like robot is doing okay.

AntonioConsilvio, 2024-06-25:
Hi @paliasgh, if moving it by hand the only encoder frozen is the pitch, it is probably the cause of the HW faults.

Sometimes it works sometimes it doesn't, probably because the connector (or crimp) is about to disconnect, but it isn't completely, so sometimes it works sometimes it doesn't.

The left shoulder pitch sensor is connected to an electronic board called EMS 10.0.1.1 (EB1).

I don't think the problem is on the sensor, I think it's more likely that the cause is on the EMS side.

[image]

The problem is that the EMS is under the backpack, so unfortunately you'll have to remove it again 😕 to get to the connector.

[image]

Once you have reached the connector, I recommend testing it with tweezers for crimp tightness!

Please, send us feedback and if you have any questions feel free to ask!

paliasgh, 2024-07-29:
Hello @AntonioConsilvio ,

Thanks for the instruction. Since your last message, our robot has been doing fine, so we did not continue checking for the issue. Recently, the issue is happening again... 

I did another test and it looks like only manually moving Joint 2 is enough for sending the shoulder to the same HW fault. It is not clear if any encoder value freezes or not everytime, but I could see in one test that Joint 2 got stuck:

https://github.com/user-attachments/assets/51470d7c-4df9-4ce2-853f-3ac87a5055e8

Another time, however, the values kept changing even after the fault:

https://github.com/user-attachments/assets/0a478f9d-9674-4815-8dc5-420482f094f1

Do you think Joint 2 (yaw) can cause the issue instead of Joint 0? Could you please let us know how we can check this one as well? Many thanks!

AntonioConsilvio, 2024-08-09:
Hi @paliasgh! Unfortunately, if none of the joints get stuck, it's difficult to figure out which of the three sensors is the cause because they are all linked together and the same error appears on all three. 

However, they are all connected to the same board shown [here](https://github.com/robotology/icub-tech-support/issues/1822#issuecomment-2189168903), so the connectors to check are always there.

But some sensors (e.g. the ROLL or the YAW) have a male-female connector and that too may have some damaged cable.

So all these connectors should be checked.

[image]

I think it is difficult that the problem could be on the sensor as the cables are soldered (however, it is possible).

Please, send us feedback and if you have any questions feel free to ask!

AntonioConsilvio, 2025-05-07:
Hi! Closing due this comment:

- https://github.com/robotology/icub-tech-support/issues/1926#issuecomment-2857987290

closed 2025-05-07 by AntonioConsilvio