robotology/mesh-support #1823: Right wrist slightly bent

SimoneMic, 2024-06-06:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

In home position both wrists should be in the same position, if not perfectly symmethrical, at least more or less aligned.
Instead, the right wrist start slightly bent outwards. When on the yarpmotorgui all the joints references are in the same zero position.

### Detailed context

As you can see from the pictures both wrists should be in the same orientation, this means that there could be a wrong encoder reading, or an issue in the zeroing/calibration procedure.

[image]

[image]

[image]

AntonioConsilvio, 2024-06-10:
Hi @SimoneMic, the belt of one of the wrist motors broke.

[image]

We removed the lowerarm from the robot and we already started with the replacement.

cc @maggia80 @S-Dafarra @fbiggi

AntonioConsilvio, 2024-06-18:
Hi @SimoneMic, @fbiggi and @AntonioAzocar replaced the broken belt successfully. I calibrated the wrist with new values and tested the joints, It works properly now!

Here the PR with the new calibration values:

- https://github.com/robotology/robots-configuration/pull/655

Thank you everyone for the help! 🚀

cc @S-Dafarra

closed 2024-06-18 by AntonioConsilvio