robotology/mesh-support #1824: iCub board 22 face avoid to turn on yarprobotinterface

lunagava, 2024-06-11:
### Robot Name 🤖

iCubGenova02 S/N:001

### Request/Failure description

The output of running $yarprobotinterface inside icub-head is giving me this error:

### Detailed context

[WARNING]  from BOARD 10.0.1.22 (face-eb22-j0), src LOCAL, adr 0, time 711s 703m 292u: (code 0x0000000d, par16 0x017a par64 0x0029017a00fd035e) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
[ERROR]  from BOARD 10.0.1.22 (face-eb22-j0), src CAN1, adr 0, time 711s 706m 131u: (code 0x00000019, par16 0x2901 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .

valegagge, 2024-06-12:
Hi @lunagava ,
from the message you reported, it seems that the fw cannot transmit any CAN packet (TX queue full with 41 packets). I guess that there is an issue on the CAN1 cable.

cc @AntonioConsilvio

filippoborgogni, 2025-02-14:
Hi @lunagava, I have worked on this issue and I found a cable damaged (cable name: N9). I fixed it and now it works properly.
Closing!🚀

@AntonioConsilvio

closed 2025-02-14 by filippoborgogni