robotology/mesh-support #1826: The CPU Button Automatically Shuts Down After Being Pressed to Green for 60 Seconds

jimingda, 2024-06-12:
### Robot Name 🤖

iCubShanghai01 S/N:049

### Request/Failure description

When I press the CPU button, it turns green. However, after approximately 60 seconds, it turns red and shuts down automatically.

### Detailed context

Is it possible to check whether there is damages to the power board? 

### Additional context

N/A

### How does it affect you?

The iCub can't be turned on.

jimingda, 2024-06-27:
Hi,

Anyone can have a look at this issue?

Appreciate in advance.

AntonioConsilvio, 2024-07-01:
Hello @jimingda! Sorry for the late reply.

Before you can solve the problem, you have to find out the cause.

So as a first test, we suggest removing the head cover and disconnecting the CPU power connectors (this should be done with the robot switched off).

To remove the cover, unscrew the two side screws:

[image]
(The other screw is on the other side of the head)

Removing the cover will expose the power cables:

[image]

Disconnect them and then try switching on the CPU and see how it behaves!

Please, send us feedback and if you have any questions feel free to ask!

jimingda, 2024-07-02:
Thank you @AntonioConsilvio 

I removed the cover but couldn't find the white connector as shown in your picture. Below is a photo I took of the inside of the iCubShanghai's head.

[image]
[image]

AntonioConsilvio, 2024-07-02:
Hi @jimingda , you can safely remove the back cover of the head for better visibility (it should not be held with screws, just pull it).

After that, the connectors should be there, try pulling the yellow and black cables out of the head and see if they come out!

Feel free to leave me feedback afterwards!

AntonioConsilvio, 2024-07-02:
Hi @jimingda, I don't think the connectors are in the sheath connecting the back to the head.

I think they might be in the head. For example under the plate (red circle in the picture below), so try pulling the yellow and black cables out by, for example, taking them from the green circle and pulling them outwards.

[image]

jimingda, 2024-07-03:
Hi @AntonioConsilvio 

I found two connectors, one connects 2 cables another connects 3 cables, and it seems difficult to get it out.
 
[image]
[image]

AntonioConsilvio, 2024-07-03:
hi @jimingda, as shown in this [picture](https://github.com/robotology/icub-tech-support/issues/1826#issuecomment-2202681290), you can try pulling the cables from the green circle and see if the connectors come out.

If they do not come out, you can try unscrewing four screws and removing the top plate, so that the connectors are exposed and more visible.

Two screws are the ones circled in red in the photo below, the other two are on the other side of the head. (obviously with the robot switched off)

[image]

jimingda, 2024-07-03:
Hi @AntonioConsilvio 

I can pull the cables from the green circle, and disconnect the connector.
[image]

AntonioConsilvio, 2024-07-03:
Great @jimingda!

We now have 3 possible causes:

- The CPU
- The RFE board
- The face expressions

To figure out the cause, I would suggest you follow the steps below:

- Reconnect the connector and start the robot and see if the problem recurs.

- If it recurs, try disconnecting the power supply (yellow and black cable) of the board you see circled in red in the photo below (board name: `RFE`):
(you can see the cables in the photo and I have circled them in blue).

[image]

At this point only the CPU remains connected to the power supply, while the `RFE` and face expressions are disconnected.

Try switching the robot on and see how it behaves!

jimingda, 2024-07-08:
Hi @AntonioConsilvio 

Any advices on `when I pressed the Motor button, it did not turn green, and the CPU button shut down automatically as well.` ?

Thanks.

AntonioConsilvio, 2024-07-08:
Hi @jimingda, close to the power connector you disconnected (2 PINs), there should be another one (3 PINs). 
You can also see it in the photo you sent:

[image]

I would try unplugging it, so you will have disconnected the head boards that are turned on when you turn on the motors.

At this point I would advise you to switch on the CPU and then the motors to see how the robot behaves.

jimingda, 2024-07-10:
Hi  @AntonioConsilvio 

Shall i disconnect the 2pin connector at the same time when i disconnect the 3pin connector？

jimingda, 2024-07-11:
Hi @AntonioConsilvio 

What is the possible causes of the CPU power problem

jimingda, 2024-07-16:
Hi @AntonioConsilvio,

I apologize for any inconvenience, but could you please spare a moment to look into this issue? I have been struggling with it for quite some time.

Thank you!

AntonioConsilvio, 2024-07-17:
Hi @jimingda, sorry for the wait. I've been busy with some internal demonstrations over the last few days.
 
So, according to the latest information you provided, the problem seems to be with the motor power supply, which due to a short circuit, for example, could cause the robot to shut down.
We need to find out in which part of the robot this problem occurs by exclusion.
 
To do this, I would start with the power source, namely the `BDC` board. 
First, reconnect the previously disconnected head connectors. Then to reach the `BDC` board, you have to remove the backpack by unscrewing the screws and pulling it carefully:
 
[image]

[image]

At this point, the board looks like this on one side of the robot:
 
[image]

Following the diagram, I recommend that you unscrew all the alimetations (j3 - j4 - j5) and do a power-up test to see how it behaves.

[image]

After that, if the buttons don't behave strangely, I suggest to reconnect everything step by step.

First the CPU (j5), and do a power-up test.

If all goes well, try connecting the motors (j4) and test.

And finally the body motors (j3).

When the problem reappears then the problem is in that part of the robot. 

Please, send us feedback and if you have any questions feel free to ask!

AntonioConsilvio, 2024-07-19:
Hi @jimingda, you have to unscrew the eyebolts too:

[image]

jimingda, 2024-07-19:
Thank you @AntonioConsilvio  for quick response.

I still can't remove the backpack after unscrew the eyebolts. Maybe there are other connections i didn't remove.
Shall i cut this on the bottom of the backpack?
[image]

AntonioConsilvio, 2024-07-19:
Yes! Cut the cable tie

jimingda, 2024-07-21:
Hi @AntonioConsilvio 

I cut that cable tie, but still  can't remove the backpack.

AntonioConsilvio, 2024-07-22:
Hi @jimingda make sure you have removed all connections (such as power supply or Ethernet):

[image]

And try unscrewing these two screws (red in the picture below), then remove the dummy battery by pulling up on the lever circled in blue: 

[image]

Then try removing the cover.

> [!Note]
> TIP: in the picture above, you may notice a green circle, it may help to loosen (not remove) that screw, and also the one next to it that holds the same electronic board, and carefully push the board against the mechanics. By doing so, the buttons will not be too much of a nuisance when removing the cover. It is not necessary, but useful.

jimingda, 2024-07-23:
Thanks @AntonioConsilvio 

I manage to remove the backpack. Then do the test as your advice in the reverse sequence.

First, disconnect  j3 (which means j4 & j5 connected), turn on CPU and then turn on Motor, after 30 seconds, both shut down automatically. 

Then, disconnect j4 (which means only j5 connected, as the pic below), turn on CPU and then turn on Motor. It's normal now, CPU and Motor power didn't shut down. 

[image]

AntonioConsilvio, 2024-07-23:
Hi @jimingda, great job! 

Then, let's exclude J3 from the possible cause.

Let's talk instead about j4, which seems to be the problem there. 

As you can see in the electrical diagram in this [comment](https://github.com/robotology/icub-tech-support/issues/1826#issuecomment-2233229182):

At PIN1 and 4 (J4_1 and J4_4) the 12V of the head boards are connected, while at PIN2 and 3 (J4_2 and J4_3), the 12V of the arms boards are connected.

Try keeping only one of the two connected (e.g. those of the arms) and start the robot and see how it behaves. And then only the head ones and see how it behaves. By doing this test we should identify on which power supply the problem is located.

Regarding this question:

> How can i take the BDC from the backboard, because there is limited space to put the power cable into connector j3 j4 j5.

The board screws are behind the board itself, internal, where the dummy battery lodges. If you remove the dummy battery and look inside, you should see the screws.

Please, send us feedback and if you have any questions feel free to ask!

jimingda, 2024-07-23:
Thanks @AntonioConsilvio 

I connect J4_1 and J4_4 (now only J5 & J4_1 J4_4 connected), turn on CPU and then Motor, after about 60 seconds they shut down automatically.

AntonioConsilvio, 2024-07-23:
@jimingda and with J4_2 and J4_3 instead J4_1 and J4_4 what happens?

jimingda, 2024-07-23:
@AntonioConsilvio 

Since i haven't take the BDC from the backpack, the space is limited.
Can i distribute the Larm_GND & Rarm_GND in J4_3 and J4_4, and distribute the Larm_12V & Rarm_12V in J4_1 and J4_2,
or just follow the diagram put two cables in one hole?

AntonioConsilvio, 2024-07-24:
Hi @jimingda, The idea is to keep as close to schematic as possible (also during debugging). 
However, if you have huge difficulties, you can do what you said:

> Can i distribute the Larm_GND & Rarm_GND in J4_3 and J4_4, and distribute the Larm_12V & Rarm_12V in J4_1 and J4_2,
> or just follow the diagram put two cables in one hole?

jimingda, 2024-07-24:
@AntonioConsilvio 

I connected J4_2 and J4_3 (now only J5, J4_2, and J4_3 are connected), turned on the CPU and then the motor. After about 60 seconds, they shut down automatically too. Since I haven't installed the left arm, only the right arm is functioning now.

AntonioConsilvio, 2024-07-25:
Hi @jimingda, unfortunately I will not be available for several days, when I return we will try to figure out how to proceed!

Here you can find the electronics diagram of the board:

- https://github.com/icub-tech-iit/electronics-boards-public/tree/master/power-management/backpack-bdc

jimingda, 2024-08-12:
Thanks @AntonioConsilvio 

I disconnected the J4 cable and checked the voltages with a multimeter after powering on the CPU and Motor. The voltage was 12V. Then, I connected the head cable (+12V and GND), powered on the CPU, and then the Motor. The voltage remained at 12V, but it shut down automatically after 60 seconds, just like before.

jimingda, 2024-08-22:
Hi @AntonioConsilvio 

It appears that the BDC board is functioning correctly and delivering the necessary voltage. Do you have any insights into what might be causing the issue?

AntonioConsilvio, 2024-08-27:
Hi @jimingda! Have you tried to test whether there is continuity (and thus a short circuit) between 12V and GND, as suggested here:

> You could try disconnecting all the wires from J4, and using a multimeter to check if the +12V and GND are short circuited.

jimingda, 2024-09-11:
Hi @AntonioConsilvio 

I disconnected all the wires on J4. Before powering up, I verified that there was no short circuit between +12V and GND.

Then, I powered the CPU button and tested the voltage between +12V and GND with a multimeter. I found that there was a short circuit on J4.

jimingda, 2024-09-19:
Hi @AntonioConsilvio 

Where can i find the schematic diagram of BDC board, i check the docs and didn't find it.
https://github.com/icub-tech-iit/electronics-boards-public/tree/master/power-management/backpack-bdc/docs

Thanks in advance.

AntonioConsilvio, 2024-09-27:
Hi! Sorry for the long wait, I am aware of your issue, but unfortunately we do not have the availability to provide the proper support at the moment. We will get back to you as soon as possible.

jimingda, 2024-11-29:
Hi @AntonioConsilvio,

The issue seems to be resolved. The root cause was that the current limit on the power transformer was set to 1A, which was insufficient to power the motor, causing it to shut down automatically. After increasing the current limit to 10A, the problem no longer occurred.

I know the voltage is set to 40V, but what is the recommended current limit for iCub? I couldn’t find this information in the documentation.

Thanks for your attention!

closed 2024-12-05 by jimingda