robotology/mesh-support #1828: Right arm FT sensor is saturated and does not work properly

gabrielenava, 2024-06-12:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

While debugging we noted that the right arm FT sensor does not provide the right values despite it starts correctly during startup. Actually, it seems that all forces and moments provided by the FT are constant and saturated to big values.

### Detailed context

See e.g. this plot of the FT readings

[image]

### Additional context

cc @davidegorbani @HosameldinMohamed @DanielePucci

HosameldinMohamed, 2024-06-13:
This is the same F/T sensor that we replaced on https://github.com/robotology/icub-tech-support/issues/1622.

The sensor was "almost" performing as expected until we found this issue.

Interestingly, this seems to be the same issue of the older F/T sensor that was replaced before.

AntonioConsilvio, 2024-07-01:
Hi @gabrielenava @HosameldinMohamed, we replaced the FT sensor (code: 12487.A3 SN: 095) with a new one code: 12487.A5 SN: 129)! 

Please give us feedback on how it works!

gabrielenava, 2024-07-05:
hello @AntonioConsilvio, I have tested the FT sensor and it seems it is working fine! Thank you again!

AntonioConsilvio, 2024-07-05:
Very nice! Closing 🚀