robotology/mesh-support #1830: Left shoulder roll overcurrrent

evelyd, 2024-06-18:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The left shoulder roll failed two separate times in the span of an hour.

### Detailed context

The error it gives is that the i2c limit is exceeded, and we saw in the logged data that the current for the left shoulder roll motor is much higher than that of the right shoulder roll. Touching the motor shows that it is very hot.

@S-Dafarra mentioned that there was a similar issue during the demo last week in Bologna.

### Additional context

[image]

### How does it affect you?

I can't test walking trajectories without risking burning this motor.

maggia80, 2024-06-18:
thanks @evelyd, could you please add the log of the yarprobotinterface?

AntonioConsilvio, 2024-06-18:
Hi! I would like to add the interventions that we did in Bologna to try to solve the problem (clearly not enough):

- 2FOC replacement

- LCORE5 replacement

- Optical disc cleaning

The robot worked correctly during the demo, but the problem reoccurred when we returned.

evelyd, 2024-06-18:
> thanks @evelyd, could you please add the log of the yarprobotinterface?

Hi, unfortunately I don't have that log. But here is the logged data from the last test before the shoulder faulted (the fault happened while the robot was just hanging in the air afterward): [mat file](https://istitutoitalianotecnologia-my.sharepoint.com/:u:/g/personal/evelyn_delia_iit_it/EaKjgFbz9qtEnP7Jphlypi8B4wWtLq3mDET_2FPYJbJ_jw?e=BBWuyG) in case it helps.

evelyd, 2024-06-19:
Hi, as an update, I tried to use the robot again today, and after running the calibration this motor kept faulting, so it seems like it's burned.

Here is the yarprobotinterface log from this event @maggia80: 
[l_shoulder_roll_fault_log.txt](https://github.com/user-attachments/files/15898031/l_shoulder_roll_fault_log.txt)

isorrentino, 2024-06-20:
Today I started the `yarprobotinterface` with the modifications introduced by @AntonioConsilvio to recalibrate the motor

[image]

but the motor went in the opposite direction and the arm broke the neck tendons.

Here is the log

[log_ergocub-torso_yarprobotinterface_7194.txt](https://github.com/user-attachments/files/15909802/log_ergocub-torso_yarprobotinterface_7194.txt)

The next two runs the shoulder pitch went to hardware fault and the shoulder roll did not calibrate.

[log_ergocub-torso_yarprobotinterface_12082.txt](https://github.com/user-attachments/files/15909879/log_ergocub-torso_yarprobotinterface_12082.txt)

AntonioConsilvio, 2024-06-21:
Hi! @fbiggi opened the coupling and cleaned the motor and gearbox.

Also replaced the optical disc and then recalibrated the joint, see PR:

- https://github.com/robotology/robots-configuration/pull/658

This intervention was not enough, so we tested the motor phases and they were fine. We also tried to move the motor with a motor test tool and it worked properly.

Then we tried testing the LCORE5 and it was giving some problems. So I tested the cables from the connector that came out of the crimp. 

Now I fixed the wiring, and put a new cover on the sensor to prevent it from getting damaged again.

I tested the robot for a while and it seems to be working properly!

I leave the issue still open for any feedback.

cc @maggia80 @evelyd @isorrentino @S-Dafarra

AntonioConsilvio, 2024-07-22:
Since the problem did not reappeared, I proceed with closing the issue!

closed 2024-07-22 by AntonioConsilvio