robotology/mesh-support #1833: Left shoulder pitch in HF when starting the robot

isorrentino, 2024-06-20:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The left shoulder pitch does not calibrate

### Detailed context

Here the logger 

[log_ergocub-torso_yarprobotinterface_15692.txt](https://github.com/user-attachments/files/15910157/log_ergocub-torso_yarprobotinterface_15692.txt)

maggia80, 2024-06-20:
`1486.102193 <ERROR> from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=10s 904m 320u : CFG: EOtheEncoderReader cannot be configured: verification fails joint 0 (l_shoulder_pitch) has error on primary encoder (code=9). `

isorrentino, 2024-06-25:
The shoulder pitch went again to the wrong initial position, broke the neck and now does not start.

[log_ergocub-torso_yarprobotinterface_5282.txt](https://github.com/user-attachments/files/15965640/log_ergocub-torso_yarprobotinterface_5282.txt)

AntonioConsilvio, 2024-06-25:
Hi! The malfunction is probably caused by the AMO sensor.

The cable was slightly damaged on the connector side on the EMS. For now the AMO is disabled and only LCORE5 is used. We will soon fix the cable and re-enable the sensor.

With LCORE5 the joint is working properly.

S-Dafarra, 2024-06-27:
I pushed the configuration in https://github.com/icub-tech-iit/robots-configuration/commit/2883e9c993b48f4ea6ee099bfea194a0639f5b2e

AntonioConsilvio, 2025-02-14:
Hi, this issue was due a damaged cable that connect the AMO sensor to the EMS EB2. I fixed this problem, closing! ✅

closed 2025-02-14 by AntonioConsilvio