robotology/mesh-support #1836: Issue with pinky of the right arm

ddetommaso, 2024-06-21:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

During calibration the right hand take a while to start closing fingers and then, during the closing, the fingers get stuck

### Detailed context

similar to previous issue #1805 

### Additional context

[log_icub-head_yarprobotinterface_1846.txt](https://github.com/user-attachments/files/15924301/log_icub-head_yarprobotinterface_1846.txt)

### How does it affect you?

We have an important demo on the 10th of July. If possible please we would ask you to schedule the intervention between the 2nd and the 5th of July. As soon as you schedule the intervention please send a notification to us.

ddetommaso, 2024-06-26:
My colleague Mateusz will be the reference person in Erzelli in the next following days cc @mmwozniak

AntonioConsilvio, 2024-07-01:
Check the comment:

- https://github.com/robotology/icub-tech-support/issues/1841#issuecomment-2199739173

AntonioConsilvio, 2024-07-05:
Closing due this comment:

- https://github.com/robotology/icub-tech-support/issues/1841#issuecomment-2210334608

closed 2024-07-05 by AntonioConsilvio