robotology/mesh-support #1837: Neck broken due to the wrong initial calibration of left shoulder pitch

isorrentino, 2024-06-25:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The shoulder pitch went to the wrong initial configuration and broke the neck

### Detailed context

Here the exported log.

[log_ergocub-torso_yarprobotinterface_5282.txt](https://github.com/user-attachments/files/15965742/log_ergocub-torso_yarprobotinterface_5282.txt)

cc @simonemic

SimoneMic, 2024-06-25:
Hello,

If the neck could be repaired this morning we can use the robot for the demo (of the open day) this afternoon.
Otherwise I'll use the 002, hopefully the demo run smoothly on that robot at the first trial.

Thanks 🙏

isorrentino, 2024-06-25:
They are repairing the neck now :)

AntonioConsilvio, 2024-06-25:
Hi @isorrentino @SimoneMic, me and @fbiggi replaced the broken tendon with a new one, we tested the robot and it works properly now! 

Here the issue for the wrong shoulder pitch calibration:

- https://github.com/robotology/icub-tech-support/issues/1833

Closing! ✅

closed 2024-06-25 by AntonioConsilvio