robotology/mesh-support #1838: Left ankle roll Hardware limit pin detached from the mechanical structure

GiulioRomualdi, 2024-06-25:
### Robot Name 🤖

ergoCub 1.1 S/N:002

### Request/Failure description

Today @S-Dafarra tried the walking controller and noticed that the robot was not able to walk anymore. 
After some tests we noticed that the problem was related  to the left ankle roll pin (the one used for the calibration that was detached)

### Detailed context

Here is the picture of the PIN

### Additional context

@AntonioConsilvio glued the pin. We recalibrated the robot and we tested the walking. Everything was working as expected (PR to be opened)
```patch
diff --git a/ergoCubSN002/calibrators/left_leg-calib.xml b/ergoCubSN002/calibrators/left_leg-calib.xml
index 9433981ff..e7a78ceeb 100644
--- a/ergoCubSN002/calibrators/left_leg-calib.xml
+++ b/ergoCubSN002/calibrators/left_leg-calib.xml
@@ -20,7 +20,7 @@
     <param name="calibration3">                       0.0            0.0          0             0          0       0        </param> 
     <param name="calibration4">                       0.0            0.0          0             0          0       0        </param>
     <param name="calibration5">                       0.0            0.0          0             0          0       0        </param>
-    <param name="calibrationZero">                    109.27         110.848      81.65         6.38       -46.624 25.808   </param>
+    <param name="calibrationZero">                    109.27         110.848      81.65         6.38       -46.624 27.187   </param>
     <param name="calibrationDelta">                   0              0            0             0          0       0        </param>
     <param name="startupPosition">                    90             80           0             -95        0       0        </param>
     <param name="startupVelocity">                    10.0           10.0         10.0          10         10      10       </param>
```

AntonioConsilvio, 2024-06-28:
Hi! the hole where the hardware limit is mounted has become ovalised and therefore the hardware limit is no longer stationary.

[image]

We are now re-working the part so that we can fix the hardware limit so that it works properly again.

AntonioConsilvio, 2024-09-16:
Hello, the replacement part has arrived, but unfortunately there was a manufacturing error and the part will not fit on the robot.

As the robot is working properly, we will close this support issue and monitor the issue internally.

closed 2024-09-16 by AntonioConsilvio