robotology/mesh-support #1839: The fingers may refuse to close completely until a full reboot

S-Dafarra, 2024-06-26:
### Robot Name 🤖

ergoCub 1.1 S/N:002

### Request/Failure description

During a demo in Washington, we noticed that the left pinky finger was not moving more than 8 degrees. There was no particular issue, with the mechanics or the electronics, but it simply would not close, neither via the ``yarpmotorgui``.

### Detailed context

Once back home, @AntonioConsilvio tried to move the left pinky again and this time there was no issue. Most likely it was enough to reboot, but still, it is weird that at a certain point, the fingers decide to stop moving.

AntonioConsilvio, 2024-08-06:
Hi! Regarding the pinky of the left hand, the problem described in this issue resulted in the tendon rupture.

I replaced the tendon and the problem did not reoccur, however I would keep the issue open waiting for more extended tests.

AntonioConsilvio, 2024-08-29:
Hi! Closing due this comment:

- https://github.com/robotology/icub-tech-support/issues/1857#issuecomment-2317428691

closed 2024-08-29 by AntonioConsilvio