robotology/mesh-support #1840: Joint legs miscalibrated

GiulioRomualdi, 2024-06-26:
### Robot Name 🤖

ergoCub 1.1 S/N:002

### Request/Failure description

Today with @LoreMoretti we tried to run the walking controller and we noticed that the robot was hardly impacting the ground making the robot falling 

### Detailed context

Asking for the joints to move the legs in zero configuration we noticed that the feet are not parallel with respect to the ground

### Additional context

Yesterday @AntonioConsilvio fixed https://github.com/robotology/icub-tech-support/issues/1838  we changed the roll ankle calibrator. Reverting did not solve the issue. Indeed there seems to be a problem with the pitches (we dont know if it is the ankle pitch,  hip pitch or knee) 

### How does it affect you?

We cannot make the robot walk in a reliable way. Making all the experimental activity blocked
A demo that could be impacted by this issue:  
- https://github.com/ami-iit/lab-events-demos/issues/324

AntonioConsilvio, 2024-07-01:
Hi! after the temporary fix of the ankle roll (see [issue](https://github.com/robotology/icub-tech-support/issues/1838)).

Together with @GiulioRomualdi we re-performed a fine calibration and the robot started working and walking properly again! ✅

AntonioConsilvio, 2024-07-02:
Since with the latest calibration the robot works and walks properly, I am closing the issue.

If the problem occurs again, feel free to open a new one! @GiulioRomualdi

closed 2024-07-02 by AntonioConsilvio