robotology/mesh-support #1841: Right shoulder tendon broken

ddetommaso, 2024-06-26:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

one of the tendon broke and is out of its pulley

### Detailed context

[image]
[image]

### How does it affect you?

We have participants scheduled for the next 2 following days and we might need to cancel them and reschedule them as soon as the robot is fixed. Additionally, we have an important demo on the 10th of July. If possible please we would ask you to schedule the intervention as soon as possible. As soon as you schedule the intervention please send a notification to us.

ddetommaso, 2024-06-26:
My colleague Mateusz will be the reference person in Erzelli in the next following days cc @mmwozniak

mmwozniak, 2024-06-26:
**How does it affect us**:
We are currently collecting data for a paper to be submitted to the Humanoids conference (deadline for finished article 15.07) and we need to finish data collection as soon as possible. Originally we planned to finish data collection this Friday

AntonioConsilvio, 2024-07-01:
Hi @mmwozniak @ddetommaso, we planned to carry out the maintenance work on Thursday 4 July!

AntonioConsilvio, 2024-07-05:
Hi @ddetommaso @mmwozniak, @fbiggi and I have replaced the broken tendon with a new one.
We tested the robot and it is now working properly! 🚀

In addition, @mmwozniak performed the experiment and it worked properly! 

Closing! ✅

closed 2024-07-05 by AntonioConsilvio