robotology/mesh-support #1842: Various fixes/improvements on ergoCubSN000

GiulioRomualdi, 2024-06-28:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

Hi all,

ergoCubSN000 requires some maintenance. After a meeting with @S-Dafarra and @DanielePucci, we prepared a list of fixes that could be implemented to improve the robot's structure and make it comparable with the SN001 and SN002.

### Detailed context

We split the list into must-have, nice-to-have, and wish categories. Here is the list:

### Must-have
1. The head is moving when controlled. I think the tendon should be tensioned/fixed.
2. Firmware needs to be updated to include all safety features already in place for the other two ergocubs (temperature checking, compliant mode fixed).
### Nice-to-have
1. If the new tendons for the neck are available, they could be mounted in place of the original ones (similar to what was done on ergoCubSN001).
2. New wrist mechanism mounted (the same as used for ergoCubSN001 and ergoCubSN002).
### Wish list 
1. New versions of the hand (the ones from ergoCubSN001 and ergoCubSN002) installed.

### Additional context

cc @Fabrizio69 @maggia80 @DanielePucci @S-Dafarra

AntonioConsilvio, 2024-08-07:
Hi @GiulioRomualdi @S-Dafarra, Activities performed (about a month ago) on the basis of available material: 

- Updating FW to the latest devel (see [commit](https://github.com/robotology/icub-firmware-build/pull/153)). ✅

- Robotology-superbuild update. ✅

- I tensioned the neck tendons as far as possible (we did not replace the tendons as we are short of them) ✅

- Changed configuration files so that temperature sensors can be enabled (here the [PR](https://github.com/robotology/robots-configuration/pull/666)). ✅

- I updated the FW of BAT (1.3.4) and OLED (1.3.1) ✅

- @AntonioAzocar successfully changed the hip pitch AMO routing so that the cables are safer! ✅

I tested the robot and it works properly! 🚀

> [!Note]
> Since it wasn't one of the required updates, I didn't touch the robotology-superbuild of the head, so I don't know if you have an interest in keeping it in its current state or just didn't think of it.

AntonioConsilvio, 2024-11-12:
Hi! With the replacement of neck tendons due to maintenance work, the robot now has new tendons with a larger diameter (0.9 mm). 🚀

At this time there are no plans for further upgrades to the robot in the near future, so I will proceed with closing this issue.

I do not exclude the possibility that there may be upgrades in the future, in that case you will be informed!

closed 2024-11-12 by AntonioConsilvio