robotology/mesh-support #1843: Neck tendon broken

evelyd, 2024-06-28:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The neck tendon snapped during locomotion testing.

### Detailed context

When this happened, the neck roll motor went into fault. We stopped the experiments following this occurrence.

### Additional context

## Neck tendon
[image]

## Log
[log_ergocub001-lap_yarpmotorgui_36744.txt](https://github.com/user-attachments/files/16034642/log_ergocub001-lap_yarpmotorgui_36744.txt)

evelyd, 2024-06-28:
@GiulioRomualdi @EhsanRanjbari  @isorrentino

AntonioConsilvio, 2024-07-02:
Hi @evelyd, @fbiggi and me replaced the broken tendon with a new one and tested the robot. It is working properly now! 

Closing! 🚀

closed 2024-07-02 by AntonioConsilvio