robotology/mesh-support #1844: Backlash on the right ankle pitch

gabrielenava, 2024-07-01:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

We noted that there is a lot of backlash when moving the right ankle pitch joint. It does not seem to be damaged at first sight, probably some internal screws got loose.

### Detailed context

https://github.com/robotology/icub-tech-support/assets/12396934/bd4885ab-2e49-4a2a-b65f-e28e6714a920

### How does it affect you?

affects robot stability during balancing for iRonCub experiments

AntonioConsilvio, 2024-09-05:
Hi! I fixed the backlash on the ankle pitch tightening the following screws (orange 🟠).

[image]

Now the backlash is gone, closing! ✅

gabrielenava, 2024-09-05:
hi @AntonioConsilvio this issue is fixed, thank you!