robotology/mesh-support #1850: Neck cable broken

rob-mau, 2024-07-10:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The neck cable broke when the robot fell over.

### Detailed context

We were doing tests with a step, using bungee cords to hold the robot and allow more dynamic movements. We accidentally stopped the motors while the robot was standing on the step, and it fell, and this broke the neck cable.

### Additional context

[image]

AntonioConsilvio, 2024-07-16:
HI @rob-mau, @fbiggi and I replaced the broken tendons and tested the robot. It is working properly now!

> [!Note]
> Both tendons were broken.

rob-mau, 2024-07-16:
Thanks a lot!