robotology/mesh-support #1852: Ubuntu Partition Not Starting On Robot Laptop

SimoneMic, 2024-07-16:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

As per title, the robot partition doesn't start and was giving these errors (seems a memory corruption):
[image]

I have tried accessing the windows partition and that was working properly, but after that I wasn't able to access the dual boot option, and the laptop started directly to windows.
This is another error that shows at startup:
[image]

### Detailed context

The laptop has been used in the morning and nothing strange was reported, everything was running fine.

### How does it affect you?

The robot cannot be used.

AntonioConsilvio, 2024-07-16:
Hi @SimoneMic, @davidelasagna has restored dual boot access and it is now possible to access ubuntu.

However ubuntu doesn't boot correctly because the laptop has full storage.

Again with @davidelasagna's help we investigated a bit and found that the folders taking up the most space are:

[image]

Mostly because there are very heavy folders (tens of GB) on the desktop, probably related to data collection (or similar), or what appear to be repositories. 

> [!Important]
> I would ask you to remove anything from these folders that is not strictly necessary and takes up a lot of space.
> And I also ask you in future to be very careful when downloading or accumulating heavy files, because the robot is not usable due to the malfunction of the laptop.

cc @S-Dafarra @GiulioRomualdi

AntonioConsilvio, 2024-07-22:
I see that the PC is now working properly, thanks for freeing up some disk space.

Closing!

closed 2024-07-22 by AntonioConsilvio