robotology/mesh-support #1855: Neck tendon exited the guide

gabrielenava, 2024-07-17:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

During iRonCub experiments one of the tendon  of the neck joints exited the guide.

### Detailed context

it is the joint in the picture. Would it be possible to fix it? otherwise the head tilts during the experiments and may cause more damages
[image]

cc @DanielePucci 

### How does it affect you?

head tilts during iRonCub experiments and may cause more damages

AntonioConsilvio, 2024-07-22:
Hi @gabrielenava, I replaced the damaged tendon with a new one.

Please, give me a feedback when you know if it works properly!

gabrielenava, 2024-07-22:
thanks @AntonioConsilvio, this problem is fixed and the neck works now!

closed 2024-07-23 by AntonioConsilvio