robotology/mesh-support #1857: Hands issues

giotherobot, 2024-07-19:
### Robot Name 🤖

ergoCub 1.2 S/N:002

### Request/Failure description

There are 2 main issues in the hands:
1) In the left hand the ring/pinky fingers are not movin at all;
2) In the right hand the index and middle fingers are unable to fully close;
 
In addition in both hands the thumb is loose or falling off.

### Detailed context

# Left hand status
## Robot
[image]
## GUI
[image]

# Right hand status
## Robot
[image]
Thumb detail:
[image]
## GUI
[image]

### Additional context

Possible related issues:

https://github.com/robotology/icub-tech-support/issues/1839
https://github.com/robotology/icub-tech-support/issues/1849

### How does it affect you?

Very difficult to perform manipulation experiments and data collection.

S-Dafarra, 2024-07-22:
Now also the right pinky refuses to close, even after a full reboot.

S-Dafarra, 2024-07-24:
Despite the intervention of @AntonioConsilvio, after about 30 minutes of usage, the right pinky stopped working again

AntonioConsilvio, 2024-08-06:
Hi @giotherobot @S-Dafarra! 

## Left Hand

The problem was related to this issue:

- https://github.com/robotology/icub-tech-support/issues/1839

I replaced the tendon and now it is working. (more info check the [comment](https://github.com/robotology/icub-tech-support/issues/1839#issuecomment-2270907337)).

## Right Hand

Regarding the index and middle fingers, I was able to adjust the tendons so that they are a little tighter and improve the closure, but they still don't reach full closure.

For the time being (more for the index finger which is the worst), the only solution is to try to replace the tendon.

## Pinky problem

The problem on the right pinky, was very much the same as on the left. So I think they might have the same cause.

In fact, investigating further I tried to find out what was happening on the right pinky. Moving the joint with the jogger I noticed that after a while of use this was happening (more and more often):

https://github.com/user-attachments/assets/29b1068c-d67f-46b1-88ec-0c2b030317e4

The tendon creates a loop and prevents the finger from working properly and the tendon from breaking. I am not sure what this behaviour is due to, because I have checked the sheath and insertion and found no problems.

I have replaced the tendon and it seems to be working properly.

We will wait for a longer test to see if the problem recurs

## Thumb

To solve the problem of the thumbs coming off, the finger is being redesigned, as soon as we have the new parts they will be mounted.

AntonioConsilvio, 2024-08-29:
Hi @S-Dafarra @giotherobot, We performed numerous tests and made some mechanical modifications. Now the finger clasp should reach (or come much closer to) the hardware limit, and the tendons should no longer break so easily. 

We also fixed the thumb, so it shouldn't come off anymore. 

I will proceed with closing this issue, so that if there are future issues, we know they are after these modifications.

closed 2024-08-29 by AntonioConsilvio