robotology/mesh-support #1858: Left Shoulder Pitch not calibrating

LoreMoretti, 2024-07-22:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The left shoulder pitch does not calibrate.

### Detailed context

I have launched the calibration multiple times, but the same issue persists.

### Additional context

[image]

[left_shoulder_pitch_not_calibrating.txt](https://github.com/user-attachments/files/16330905/left_shoulder_pitch_not_calibrating.txt)

AntonioConsilvio, 2024-12-17:
Hi @LoreMoretti, is the problem still present on the robot?

LoreMoretti, 2024-12-18:
Hi @AntonioConsilvio, the problem seems solved. Thanks!

closed 2024-12-18 by AntonioConsilvio