robotology/mesh-support #1859: SKIN not working

giotherobot, 2024-07-22:
### Robot Name 🤖

ergoCub 1.2 S/N:002

### Request/Failure description

The hand fingertip skins are not working.

### Detailed context

With @S-Dafarra we checked more #1831 and found the following:

1) for the right skin the the mtb4 board is not connected or not detected, we checked using the firmware-updater:
[image]

2) removing the right skin references in `skin.xml` we were able to start the `yarprobotinterface` using that file. But the skins do not seem to be working, trying to connect to `/icub/skin/left_hand` we have a long list of 200.0 that do not react to touching the fingers at all. Here is the interface log:

[log_ergocub-torso_yarprobotinterface_4176_only_left_skin.txt](https://github.com/user-attachments/files/16333638/log_ergocub-torso_yarprobotinterface_4176_only_left_skin.txt)

AntonioConsilvio, 2024-08-06:
Hi! I probably found the cause of the problem. In fact, the 3N2 cable connector (CAN cable that connects MTB4 to EMS) on the EMS side (EB3) had the wrong colour sequence. 

The green and white were swapped.

I have fixed the wiring, when the robot upgrades are completed I will proceed with the test.

simeonedussoni, 2025-01-13:
this issue is still open? I think the skin is fixed, apart from possible cabe disconnections.

giotherobot, 2025-01-14:
I just tested the skins on the robot, there are still some issues:

1) the base name of the skins seems to be `/ergoCubSN002`, instead all the other ports of the robot have the `/ergocub` base name, not sure if this is only a configuration setting
2) the following skins are not working: left ring, left pinky, right thumb, right middle

simeonedussoni, 2025-01-15:
> I just tested the skins on the robot, there are still some issues:
> 
>     1. the base name of the skins seems to be `/ergoCubSN002`, instead all the other ports of the robot have the `/ergocub` base name, not sure if this is only a configuration setting
> 

yes, this is just a configuration setting.
if needed, you can simply edit the configuration files to make the prefix as you want and commit the modifications. 

>     2. the following skins are not working: left ring, left pinky, right thumb, right middle

fior this I need to check the cabling, probably they detached. it's a 10 minutes operation , I just need the robot accessible. feel free to drop a message on teams to meet and fix this