robotology/mesh-support #1862: Robot doesn't turn on after emergency faulting

SimoneMic, 2024-07-24:
### Robot Name 🤖

ergoCub 1.2 S/N:002

### Request/Failure description

As per title, the robot PCs, and the monitor on the back of the torso, don't turn on while powered on.
This happened after pressing the fault button.

### Detailed context

[image]
The internal led turns on, but everything else is off and it's not possible to turn on the motors and PCs

### Additional context

Cc @pas

AntonioConsilvio, 2024-07-25:
Hi @SimoneMic, the BAT board (the board that manages the power supplies for the whole robot) has stopped working. It's probably burnt out.

So I proceeded with the replacement. The new board worked for about ten minutes until it burnt out like the first one.

At the moment we don't know what causes the boards to burn out, and we have to try to figure it out. The debbugging may require some time.

> [!Note]
> From 1 August, the robot will no longer be available until the second half of August, as we will be making upgrades.  So if the BAT problem is not resolved before the upgrades start, the robot may not be available in the short period.

cc @maggia80 @Fabrizio69 @S-Dafarra @GiulioRomualdi

SimoneMic, 2024-07-25:
Thanks for the update!

AntonioConsilvio, 2024-12-18:
Hi, the problem was due to high temperatures burning out the BAT micro.

After numerous tests (and the help of @fgarini) we figured out how to lower the temperatures and thus prevent the board from burning out.

In addition, another problem had arisen, namely that in the event of short circuits on the fault line, current was reaching the BAT which would burn out the micro. To prevent this from happening again, we rewired the fault line so that would not pass through the BAT.

Closing! ✅🚀

closed 2024-12-18 by AntonioConsilvio