robotology/mesh-support #1864: `realSense` device stops streaming images when running teleoperation

mebbaid, 2024-07-26:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The realsense stops streaming images when running teleoperation. 

### Detailed context

Today we tried to test the teleoperation on the SN000. At some point the robot stops sending rgb images and the operator view is stuck. I tried to read the `/depthCamera/rgbImage:o` port directly which showed no data being sent. We restarted the device, and again after a couple of minutes of teleoperation the streaming crashes. 

I tried running the device from terminal inside the head and it throws this error
[realsense.txt](https://github.com/user-attachments/files/16390945/realsense.txt)

maggia80, 2024-07-26:
@mebbaid I remember we updated the FW of the realsense camera at some point. Could it be that ergoCub SN00) still doesn't have the upgraded FW? 
Otherwise is an hardware issue

S-Dafarra, 2024-07-26:
Thanks @maggia80! We just did that. The version was 5.14, now we updated to 5.16 following https://github.com/robotology/icub-tech-support/issues/1797#issuecomment-2102842716

We tried to move the camera tilt from one limit to the other for a minute or so and the camera feed was stable. We can check if it keeps happening

S-Dafarra, 2024-08-28:
I have been told this issue occurred again recently during some experiments run by @mebbaid and @illkid3. I am not sure about the context, but it might be related.

S-Dafarra, 2024-09-13:
cc @fedeceola @SimoneMic

fedeceola, 2024-09-14:
Hi, this issue with the realsense occurred also to us while running some experiments.

AntonioConsilvio, 2024-09-16:
Hi! I will check the Realsense hardware connection soon.

steb6, 2024-09-18:
I wanted to add that this happened to me while doing teleoperation, but not with the usual perception demo

AntonioConsilvio, 2024-09-19:
Hi, today with the help of e we checked the hardware connection. 

Maybe there could be a loose connection of the USB-C on the Xavier side.

We inserted the connector better and checked that it worked for about 1 hour.

Everything seemed to work properly, however leave me feedback regarding the RealSense operation when you try it out.

cc @mebbaid @S-Dafarra @steb6 @fedeceola @SimoneMic

S-Dafarra, 2024-09-19:
It is still happening unfortunately

S-Dafarra, 2024-09-25:
@AntonioConsilvio applied some more padding to keep the connector stable.

Moreover, I realized that we were using a very old ``librealsense2`` SDK compiled from source. I have uninstalled it and installed a more recent SDK version following https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

Now the connection seems stable. We have been using for some hours continuously and we did not notice issues.

Probably this is good to go

filippoborgogni, 2024-10-11:
Hi! We replaced the USB cable that connects the realSense to the Xavier.
The old cable didn't seems damaged, but we replaced it to make a try.
Let us know if the problem happens again!
Cc @S-Dafarra @mebbaid @steb6 @AntonioConsilvio

S-Dafarra, 2024-10-29:
Unfortunately this occurred also during the ECCV demo

AntonioConsilvio, 2024-10-29:
Hi @S-Dafarra, just to clarify, the intervention mentioned by @filippoborgogni was performed after ECCV!

S-Dafarra, 2024-10-30:
@SimoneMic did this happen during Festival della Scienza by any chance?

SimoneMic, 2024-10-30:
No @S-Dafarra, we had flawless operation from the realsense for three days.

AntonioConsilvio, 2025-03-18:
Closing due this comment:

- https://github.com/robotology/icub-tech-support/issues/2062#issuecomment-2732074078

closed 2025-03-18 by AntonioConsilvio