robotology/mesh-support #1866: Left shoulder yaw goes immediately to overcurrent

S-Dafarra, 2024-07-29:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

While teleoperating, the left shoulder yaw went into hardware fault and it was not possible to recover. It goes continuously in overcurrent.

### Detailed context

Here the full log: 
[left_shoulder_yaw.txt](https://github.com/user-attachments/files/16416923/left_shoulder_yaw.txt)

The encoders seem to be moving normally

Gandoo, 2024-07-31:
[image]

Gandoo, 2024-07-31:
> Maybe we could use some glue to seal that screw and avoid it from happening again?

ciao @S-Dafarra ...This screw has been placed with threadlocker for safety.

AntonioConsilvio, 2024-08-05:
Great job! 

Closing! 🚀

closed 2024-08-05 by AntonioConsilvio