robotology/mesh-support #1867: Robot does not turn on

LoreMoretti, 2024-07-30:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The robot does not turn on.

cc @maggia80 @isorrentino 

### Detailed context

Few seconds after connecting the robot to the power, it turned off (i.e. all displays and LEDs were turned off). Most likely it was still  absorbing power, since after a bit we saw smoke coming out of the torso and we smelled burning.

### How does it affect you?

The robot cannot be used.

Gandoo, 2024-07-31:
è stata sostituita la scheda 4B0 perché danneggiata la parte del motore M2 rotazione spalla.
il robot è in arena, eseguita l'accensione per vedere se ci fossero ancora problemi di fuori uscita di fumo sembrerebbe ok.

[image]

cc @CarlottaSartore @maggia80 @DanielePucci @Fabrizio69

Gandoo, 2024-07-31:
ciao, durante l'accensione la scheda Batt dava subito alimentazione ai motori anche se non accesi e il pulsante rimaneva lampeggiante verde. Ho smontatato la scheda e sostituito un mosfet che era in corto, ora è stata ripristinata.

[image]

[image]

cc @S-Dafarra  @CarlottaSartore @maggia80 @DanielePucci @Fabrizio69 @isorrentino

AntonioAzocar, 2024-08-01:
Hi!
The [BAT board](https://github.com/robotology/icub-tech-support/issues/1867#issuecomment-2260365975) kept giving a lot of problems due to the [2FOC](https://github.com/robotology/icub-tech-support/issues/1867#issuecomment-2260056608) burning out and I had to replace it again because there was another short circuit in this one.
Now with the new BAT board mounted it works properly!

cc @maggia80 @S-Dafarra

AntonioConsilvio, 2024-08-05:
Good job!

Closing! ✅

closed 2024-08-05 by AntonioConsilvio