robotology/mesh-support #1868: Right shoulder cables got disconnected

gabrielenava, 2024-08-02:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

During an experiment, the robot had a sudden motion with the right arm and some cables got disconnected from the crimp. See also the picture below.

### Detailed context

[image]

cc @AntonioAzocar

AntonioAzocar, 2024-08-02:
I fixed the two broken cables of the right shoulder and now i delivered the robot to @gabrielenava  for testing.

[image]

gabrielenava, 2024-08-02:
I have tested the robot and everything works now! I can move the joints in the yarpmotorgui. The problem is fixed, thank you!

AntonioConsilvio, 2024-08-05:
Good job @AntonioAzocar! Closing 🚀

closed 2024-08-05 by AntonioConsilvio