robotology/mesh-support #1870: The ergoCub head reboots automatically after shutdowm

S-Dafarra, 2024-08-07:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

When we run ``sudo poweroff`` on the ``ergocub-head`` it seems to shut down, but then it turns itself back on immediately after.
We need to shut down using the button in order to keep it off.

### Detailed context

We noticed this because the screen turned back on after a few minutes we run the ``poweroff`` command. We tried checking ``journalctl`` but we could not notice anything weird.

cc @davidelasagna

AntonioConsilvio, 2025-01-29:
Hi @S-Dafarra, we checked together with @davidelasagna and the reason is an hardware configuration.
As you can see in the [connecttech datasheet](https://connecttech.com/ftp/pdf/AGX101_Rogue.pdf) there is a switch that keeps turning the xavier on when powered. However if we put that switch in off we need to add a button connected to the carrier to press each time to turn the xavier off/on.

[image]

S-Dafarra, 2025-01-29:
Thanks @AntonioConsilvio and @davidelasagna. What is the best procedure to shut down the robot then? Should we avoid shutting down the head from Ubuntu and simply remove power?

AntonioConsilvio, 2025-01-29:
> Should we avoid shutting down the head from Ubuntu and simply remove power?

You can turn off from Ubuntu, but it will be a bit useless. So yes, the only way in this moment is simply remove the power.

AntonioConsilvio, 2025-02-20:
Talking to @S-Dafarra, we'll leave the robot like this for now and we will evaluate in the future whether we actually need to make this hardware change. Closing!