robotology/mesh-support #1872: Unable to access apt-repository from robot torso PC

SimoneMic, 2024-08-13:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

It seems that the firewall is blocking the connection from the robot torso PC to the ubuntu servers, giving a permission error on the IP.
I have checked the file `/etc/apt/sources.list` and tried to restoring the same configuration on the laptop (which is working fine), but it still isn't working.
See below for the error print:

### Detailed context

Here's the error that is given when running, as example, `sudo apt update`:
```
ergocub@ergocub-torso:~$ sudo apt-get update
Err:1 http://packages.ros.org/ros2/ubuntu jammy InRelease
  403  Forbidden [IP: 93.147.161.154 80]
Err:2 http://www.icub.eu/ubuntu jammy InRelease
  403  Forbidden [IP: 93.147.161.154 80]
Err:3 http://it.archive.ubuntu.com/ubuntu jammy InRelease
  403  Forbidden [IP: 93.147.161.154 80]
Err:4 http://it.archive.ubuntu.com/ubuntu jammy-updates InRelease
  403  Forbidden [IP: 93.147.161.154 80]
Err:5 http://it.archive.ubuntu.com/ubuntu jammy-backports InRelease
  403  Forbidden [IP: 93.147.161.154 80]
Err:6 http://it.archive.ubuntu.com/ubuntu jammy-security InRelease
  403  Forbidden [IP: 93.147.161.154 80]
```

cc @davide

S-Dafarra, 2024-09-25:
It is probably due to the setup of the proxy that was needed to access internet in Monte Porzio.
I have disabled the proxy and now ``apt update`` works.

cc @davidelasagna

davidelasagna, 2024-09-25:
Correct thanks @S-Dafarra

SimoneMic, 2024-10-04:
Closing then, thanks @S-Dafarra

closed 2024-10-04 by SimoneMic