robotology/mesh-support #1877: Fault button not working

mebbaid, 2024-08-16:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

During tests for teleoperation and walking we needed to fault the robot as it was behaving weirdly (see #1876 ) but the button was not functioning so we had to directly switch off the motors

### Detailed context

During tests for teleoperation and walking we needed to fault the robot as it was behaving weirdly (see #1876 ) but the button was not functioning so we had to directly switch off the motors

### Additional context

cc @giotherobot

steb6, 2024-09-16:
@mebbaid I just tested the fault button and it seems to work properly now

mebbaid, 2024-09-16:
Great! then we can close this issue.

closed 2024-09-16 by mebbaid