robotology/mesh-support #1880: PC on the base does not boot

fbrand-new, 2024-08-22:
### Robot Name 🤖

R1 S/N:003

### Request/Failure description

The pc on the robot base does not boot.

### Detailed context

I have tried connecting a monitor to the base and startup the robot and got a 'reboot and select proper boot device'. I have entered the bios and there is only a boot device option 'Micron_M5...' which was already selected.

### Additional context

@randaz81 was suggesting trying to boot from a USB drive and check the status of the disk from there. In order to do this we would need a USB hub since the robot base has only one usb port readily available on the base (and a keyboard at least is needed)

fbrand-new, 2024-08-28:
@davidelasagna and @AntonioConsilvio found out that the CMOS battery is dead.

Since replacing it involves dismantling the robot base, it might take time for proto team to intervene.
Davide suggested the following workaround:

1. Before starting the robot, connect a keyboard and a monitor to the base
2. Start the robot and press Canc/del immediately to open the BIOS
3. Enable the uefi boot
4. Restart the robot from the BIOS - do not restart it from the physical button else you have to repeat the procedure
5. Select 'ubuntu...' as the first boot option
6. Close and exit the bios

AntonioConsilvio, 2024-10-01:
Hi @fbrand-new! @AntonioAzocar @edersonferrante @fbiggi have successfully replaced the BIOS battery. Now the robot PC on the base boots properly. Closing ✅

closed 2024-10-01 by AntonioConsilvio