robotology/mesh-support #1883: Jetpack cables disconnected

gabrielenava, 2024-08-28:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

During indoor tests, some cables that connect to the fault button on the jetpack got disconnected. We would need help to solder them.

### Detailed context

they are the cables in the picture:

[image]

### Additional context

cc @davidegorbani @HosameldinMohamed @DanielePucci 

### How does it affect you?

we cannot perform iRonCub experiments

AntonioConsilvio, 2024-08-29:
Hi @gabrielenava, @Gandoo fixed this problem and and improved the positioning of the connector, which should prevent the problem from recurring.

Anyway, please give me feedback if you can!

gabrielenava, 2024-09-04:
thenk you @AntonioConsilvio and @Gandoo, we tested the electronics and works! I close the issue!

closed 2024-09-04 by gabrielenava