robotology/mesh-support #1891: External emergency button presserd error preventing full robot startup

vigisushrutha23, 2024-09-04:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

On starting up the robot with yarprobotinterface i get the following error.

[image]

### Detailed context

On starting the robot with the usual (yarprobotinterface --config ergocub.xml --enable_tags "(enable_ros2)" ) command, the robot first started up and did the calibration and then stopped with the error seen in the above screenshot. Along with @SimoneMic we tried with another emergency button and obtained the same error. Further observation by @SimoneMic of the board close to the emergency plug revealed a lose solder as seen in the videos below. 

https://github.com/user-attachments/assets/57aea933-ac6a-42bb-af71-89b58e7f284d

https://github.com/user-attachments/assets/9c688965-1eb3-4c7e-9a1e-24e348c1bebf

### How does it affect you?

As the robot does not startup, we are unable to verify if we can replicate the setup of modules on ergocubSn001 on this robot and thus cannot run experimental tests on this robot.

SimoneMic, 2024-09-04:
Maybe the jack connector configuration on the board is not optimal.
Since it's in a 90 degrees configuration, the force exerted by the male jack connector could easily damage and tear the soldering on the female connector side.

closed 2024-09-16 by AntonioConsilvio

fiorisi, 2024-09-16:
cc @marcellosavoldi @valegagge