robotology/mesh-support #1893: [yarprobotinterface not starting]

fedeceola, 2024-09-05:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

The yarprobotinterface does not start.

### Detailed context

The yarprobotinterface does not start.

We are rebuilding the robotology superbuild in the head of the robot to see if this fixes the problem.

cc @MariaLombardi @martinaxgloria 

### Additional context

I attach here the log from the logger.

[yarprunlog_05_09_2024_15_26_25 1.log](https://github.com/user-attachments/files/16894314/yarprunlog_05_09_2024_15_26_25.1.log)

### How does it affect you?

We cannot run experiments and demos.

fedeceola, 2024-09-05:
We recompiled the superbuild with @martinaxgloria. Now the robot starts. Should I close the issue or keep it open to investigate the problem?

AntonioConsilvio, 2024-09-05:
Great @fedeceola @martinaxgloria!

I think that we can close the issue as the problem has been solved.

However, if the problem occurs again, I suggest opening another issue so we can see how often it happens and possibly plan a study on it. 

Thank you!

closed 2024-09-06 by fedeceola