robotology/mesh-support #1894: Elbow Fault 0 issue

Ryo-Naohara, 2024-09-05:
### Robot Name 🤖

iCubLisboa01 S/N:004

### Request/Failure description

Icub's elbow doesn't work and appears as fault 0 in the yarpmotorgui

### Detailed context

When starting icub's yarpmotorinterface the right arm only moves the wrist, the rest of the arm doesn't move.
When inspecting it in yarpmotorgui the arm motors are in idle and work normally when started in yarpmotorgui, the elbow motor however appears as fault 0 in yarpmotorgui and the encoder is stuck at 164 and doesn't seem to move when the elbow is moved.
The wires coming from the board EB3 and board 2B1 were inspected with a multimeter and there seemed to be no broker or short-circuted wires.

### Additional context

[yarp_log_elbow.txt](https://github.com/user-attachments/files/16895852/yarp_log_elbow.txt)

Ryo-Naohara, 2024-09-05:
@plinioMoreno

AntonioConsilvio, 2024-09-27:
Hi! Sorry for the long wait, I am aware of your issue, but unfortunately we do not have the availability to provide the proper support at the moment. We will get back to you as soon as possible.

plinioMoreno, 2024-10-22:
@AntonioConsilvio, do you have time to help us with this issue. iCub has not been operational for the last month.

Looking forward to hearing from you

Ryo-Naohara, 2024-10-23:
Issue solved by replacing the AS5054 board of the motor.