robotology/mesh-support #1895: Robot doesn't start due to left wrist hardware fault

vigisushrutha23, 2024-09-06:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

When using the command YARP_FORWARD_LOG_ENABLE=1 yarprobotinterface --config ergocub.xml --enable_tags "(enable_ros2)" , the robot fails to start as there is an hardware fault on the left wrist.

### Detailed context

[image]

Upon running motorgui during startup we see the hardware fault on the left wrist showing error in the encoders.  

After a few restarts we started getting the following error as can be seen in the log.
[log_ergocub-torso_yarprobotinterface_3115.txt](https://github.com/user-attachments/files/16906248/log_ergocub-torso_yarprobotinterface_3115.txt)

### How does it affect you?

We are unable to perform experiments for an upcoming deadline.

vigisushrutha23, 2024-09-06:
Here is another log taken after another restart reproducing the HW fault on the left wrist. 
[log_ergocub-torso_yarprobotinterface_1866.txt](https://github.com/user-attachments/files/16906463/log_ergocub-torso_yarprobotinterface_1866.txt)

vigisushrutha23, 2024-09-06:
Following @AntonioConsilvio 's advice, we checked if the left wrist connectors were disconnected. With @mebbaid we made the connectionand the robot was able to start up. But one motion of the hands caused disconnection once again, as seen in the picture below. 

[image]

After few tries and pushing the cable behind the parallel mechanism of the wrist to prevent it from disconnecting, we succeeded and we were able to run experiments. I shall leave the issue open for further verification and solidifying the connection.

AntonioConsilvio, 2024-09-16:
Hi! I putted glue on the connector to avoid this from detached again. Closing! ✅

closed 2024-09-16 by AntonioConsilvio