robotology/mesh-support #1896: The cable of the neck broke

PasMarra, 2024-09-06:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The cable of the neck broke during the calibration.

### Detailed context

https://github.com/user-attachments/assets/973423a3-51f3-4af8-a489-a1e7dc644330

fedeceola, 2024-09-06:
Hi @AntonioConsilvio, we will need to use the robot quite intensively next week for the ICRA deadline (15th of September). Would it be possible to fix the robot as soon as possible?

cc @lornat75

enyen, 2024-09-07:
# Cound not start robot calbration

Following the broke down of the neck, ergocub startup cmd has been switched to 
```commandline
yarprobotinterface --config ergocub.xml --disable_tags "(disable_head)"
```
But got error message, ad unable to start
[image]

lornat75, 2024-09-10:
We need the robot for the ICRA deadline (15/9, can somebody check and see if it can be repaired? thanks @AntonioConsilvio

enyen, 2024-09-10:
yarprobointerface fail to perform calibration because one board is not detected @AntonioConsilvio 

[image]

[image]

AntonioConsilvio, 2024-09-16:
HI! Sorry for the late reply.

On Thursday I replaced the broken tendon and fixed the ETH problem.

The robot starts and works properly. Closing!

cc @lornat75 @PasMarra @enyen @fedeceola

closed 2024-09-16 by AntonioConsilvio