robotology/mesh-support #1899: Right shoulder roll Hardware Fault

carloscp3009, 2024-09-12:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Right Shoulder Roll Hardware Fault

### Detailed context

We were about to conduct some experiments and the right shoulder roll went into hardware fault immediately after the robot moved because of i2c error. We put the join on idle and move it manually and it went into hardaware fault saying it has reached the limit when it was actually far from it. 

CC. @GiulioRomualdi @mebbaid 

[image]

### How does it affect you?

I can not conduct the experiments

carloscp3009, 2024-09-12:
https://github.com/user-attachments/assets/be16ead5-4c2f-466d-84a1-03341e45f5d4

AntonioConsilvio, 2024-09-16:
Hi! I am closing this issue in favour of:

- https://github.com/robotology/icub-tech-support/issues/1901

closed 2024-09-16 by AntonioConsilvio