robotology/mesh-support #1900: BOARD 10.0.1.8 (left_leg-eb8-j3_5) could not tx frames on CAN bus. CanPort=CAN2.

FabioBergonti, 2024-09-12:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The robot does not start

### Detailed context

We are trying to start the robot but the interface crashes with this error:
``` 
	53,274621	ERROR			from BOARD 10.0.1.8 (left_leg-eb8-j3_5) time=218s 117m 106u :  SYS: EOtheCANservice could not tx frames on CAN bus. CanPort=CAN2. Size of fifo is 64

```
We tried multiple times, switch on/off the motors and the head but the error persits

### Additional context

[yarprunlog_12_09_2024_16_44_31.log](https://github.com/user-attachments/files/16981307/yarprunlog_12_09_2024_16_44_31.log)

### How does it affect you?

We cannot do experiments

AntonioAzocar, 2024-09-13:
I found the can cable **8N2** (connected to the **EMS EB8**) with the red wire that had the crimp incorrectly inserted in the connector, so I fixed the wiring and the problem was solved and the robot works properly

cc @AntonioConsilvio

AntonioConsilvio, 2024-09-16:
Good job @AntonioAzocar. Closing! ✅

closed 2024-09-16 by AntonioConsilvio